libretro/Lakka-LibreELEC #2034: Lakka v6.x Highend Cores freeze at closing from time to time (Intel iGPU HD 630)

shakkalakka, 2025-02-10:
hi, after updating from Lakka 5.0 the higher end cores like flycast, ppsspp, lrps2 are freezing the whole system when i close a game at rondom (no 3 to 5 closings without a freeze). it started happend with lakka 5.x nightly afer last stable relaese wich does not suffer this issue. My System is x64 with Intel i7 7700 and integrated graphics

i use this image: Lakka-Generic.x86_64-6.x-20250207-0af7dae.img.gz

Ntemis, 2025-02-10:
Can you provide some logs please?

shakkalakka, 2025-02-10:
[INFO] RetroArch 1.19.1 (Git 0792144)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE42 AES AVX AVX2 
[INFO] Version: 1.19.1
[INFO] Git: 0792144
[INFO] Built: Feb  8 2025
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[ERROR] Couldn't find any cloud sync driver named ""
[INFO] Available cloud sync drivers are:
[INFO] 	webdav
[INFO] 	null
[WARN] Going to default to first cloud sync driver...
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 59.94, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 1280x720.
[INFO] [Video]: Starting threaded video driver..
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 4 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 66 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Connector 3 connected: no
[INFO] [DRM]: Connector 3 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 4: (4096x2160) 4096 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 5: (4096x2160) 4096 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 6: (4096x2160) 4096 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 7: (4096x2160) 4096 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 8: (3840x2160) 3840 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 9: (3840x2160) 3840 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 10: (3840x2160) 3840 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 11: (3840x2160) 3840 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 12: (3840x2160) 3840 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 13: (3840x2160) 3840 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 14: (3840x2160) 3840 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 15: (3840x1600) 3840 x 1600, 59.963104 Hz
[INFO] [DRM]: Mode 16: (3840x1080) 3840 x 1080, 119.973778 Hz
[INFO] [DRM]: Mode 17: (3840x1080) 3840 x 1080, 59.968498 Hz
[INFO] [DRM]: Mode 18: (2560x1440) 2560 x 1440, 119.997597 Hz
[INFO] [DRM]: Mode 19: (2560x1080) 2560 x 1080, 143.855545 Hz
[INFO] [DRM]: Mode 20: (2560x1080) 2560 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 21: (2560x1080) 2560 x 1080, 119.879997 Hz
[INFO] [DRM]: Mode 22: (2560x1080) 2560 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 23: (2560x1080) 2560 x 1080, 59.939999 Hz
[INFO] [DRM]: Mode 24: (1920x1080) 1920 x 1080, 143.981445 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 26: (1920x1080) 1920 x 1080, 119.880005 Hz
[INFO] [DRM]: Mode 27: (1920x1080) 1920 x 1080, 100.000000 Hz
[INFO] [DRM]: Mode 28: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 29: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 30: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 31: (1920x1080i) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 32: (1920x1080i) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 33: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 34: (1920x1080i) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 35: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 36: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 37: (1920x1080) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 38: (1920x1080) 1920 x 1080, 24.000000 Hz
[... 259 lines cut ...]
[WARN] [udev]: Mouse grab/ungrab feature unavailable.
[INFO] [ALSA]: Using ALSA version 1.2.11
[INFO] [ALSA]: Using S16_LE sample format for PLAYBACK device "hdmi:CARD=PCH,DEV=0"
[INFO] [ALSA]: Period: 4 periods per buffer (768 frames, 3072 bytes)
[INFO] [ALSA]: Buffer size: 3072 frames (12288 bytes)
[INFO] [ALSA]: Can pause: yes.
[INFO] [ALSA]: Initialized PLAYBACK device "hdmi:CARD=PCH,DEV=0"
[INFO] [Audio]: Started synchronous audio driver.
[ERROR] [Microphone]: Failed to start microphone driver. Will continue without audio input.
[INFO] [Display]: Found display driver: "gl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Playlist]: Written to playlist file: "/storage/.config/retroarch/content_history.lpl".
[libretro ERROR] [SCEKERNEL] sceKernelWaitThreadEnd - bad thread 276
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[libretro ERROR] [SCEDISP] 80000104=sceDisplaySetFrameBuf(04000000, 480, 1, 1): invalid stride
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 01 seconds.
[INFO] [Config]: Loading config: "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Overrides]: Configuration overrides unloaded, original configuration restored.
[INFO] [Core]: No content, starting dummy core.
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.

shakkalakka, 2025-02-10:
[INFO] RetroArch 1.19.1 (Git 0792144)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz
[INFO] Capabilities: MMX MMXEXT SSE SSE2 SSE3 SSSE3 SSE4 SSE42 AES AVX AVX2 
[INFO] Version: 1.19.1
[INFO] Git: 0792144
[INFO] Built: Feb  8 2025
[INFO] =================================================
[INFO] [Input]: Found input driver: "udev".
[INFO] [Environ]: SET_PIXEL_FORMAT: RGB565.
[INFO] [Core]: Version of libretro API: 1, Compiled against API: 1
[INFO] [Core]: Geometry: 320x240, Aspect: 1.333, FPS: 59.94, Sample rate: 48000.00 Hz.
[INFO] [Audio]: Set audio input rate to: 48000.00 Hz.
[INFO] [Video]: Set video size to: 1280x720.
[INFO] [Video]: Starting threaded video driver..
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 4 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 66 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Connector 3 connected: no
[INFO] [DRM]: Connector 3 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 60.000000 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 4: (4096x2160) 4096 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 5: (4096x2160) 4096 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 6: (4096x2160) 4096 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 7: (4096x2160) 4096 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 8: (3840x2160) 3840 x 2160, 59.940098 Hz
[INFO] [DRM]: Mode 9: (3840x2160) 3840 x 2160, 50.000000 Hz
[INFO] [DRM]: Mode 10: (3840x2160) 3840 x 2160, 30.000000 Hz
[INFO] [DRM]: Mode 11: (3840x2160) 3840 x 2160, 29.970001 Hz
[INFO] [DRM]: Mode 12: (3840x2160) 3840 x 2160, 25.000000 Hz
[INFO] [DRM]: Mode 13: (3840x2160) 3840 x 2160, 24.000000 Hz
[INFO] [DRM]: Mode 14: (3840x2160) 3840 x 2160, 23.976000 Hz
[INFO] [DRM]: Mode 15: (3840x1600) 3840 x 1600, 59.963104 Hz
[INFO] [DRM]: Mode 16: (3840x1080) 3840 x 1080, 119.973778 Hz
[INFO] [DRM]: Mode 17: (3840x1080) 3840 x 1080, 59.968498 Hz
[INFO] [DRM]: Mode 18: (2560x1440) 2560 x 1440, 119.997597 Hz
[INFO] [DRM]: Mode 19: (2560x1080) 2560 x 1080, 143.855545 Hz
[INFO] [DRM]: Mode 20: (2560x1080) 2560 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 21: (2560x1080) 2560 x 1080, 119.879997 Hz
[INFO] [DRM]: Mode 22: (2560x1080) 2560 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 23: (2560x1080) 2560 x 1080, 59.939999 Hz
[INFO] [DRM]: Mode 24: (1920x1080) 1920 x 1080, 143.981445 Hz
[INFO] [DRM]: Mode 25: (1920x1080) 1920 x 1080, 120.000000 Hz
[INFO] [DRM]: Mode 26: (1920x1080) 1920 x 1080, 119.880005 Hz
[INFO] [DRM]: Mode 27: (1920x1080) 1920 x 1080, 100.000000 Hz
[INFO] [DRM]: Mode 28: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 29: (1920x1080) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 30: (1920x1080) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 31: (1920x1080i) 1920 x 1080, 60.000000 Hz
[INFO] [DRM]: Mode 32: (1920x1080i) 1920 x 1080, 59.940201 Hz
[INFO] [DRM]: Mode 33: (1920x1080) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 34: (1920x1080i) 1920 x 1080, 50.000000 Hz
[INFO] [DRM]: Mode 35: (1920x1080) 1920 x 1080, 30.000000 Hz
[INFO] [DRM]: Mode 36: (1920x1080) 1920 x 1080, 29.970100 Hz
[INFO] [DRM]: Mode 37: (1920x1080) 1920 x 1080, 25.000000 Hz
[INFO] [DRM]: Mode 38: (1920x1080) 1920 x 1080, 24.000000 Hz
[INFO] [DRM]: Mode 39: (1920x1080) 1920 x 1080, 23.976080 Hz
[INFO] [DRM]: Mode 40: (1680x1050) 1680 x 1050, 59.883251 Hz
[INFO] [DRM]: Mode 41: (1600x900) 1600 x 900, 60.000000 Hz
[INFO] [DRM]: Mode 42: (1280x1024) 1280 x 1024, 75.024673 Hz
[INFO] [DRM]: Mode 43: (1280x1024) 1280 x 1024, 60.019741 Hz
[... 920 lines cut ...]
[ERROR] [Microphone]: Failed to start microphone driver. Will continue without audio input.
[INFO] [Display]: Found display driver: "glcore".
[INFO] [Environ]: SET_INPUT_DESCRIPTORS:
[ERROR] [Environ]: GET_VARIABLE: reicast_texupscale - Invalid value.
[ERROR] [Environ]: GET_VARIABLE: reicast_texupscale_max_filtered_texture_size - Invalid value.
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_music_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_video_history.lpl".
[INFO] [Playlist]: Loading history file: "/storage/.config/retroarch/content_image_history.lpl".
[INFO] [Playlist]: Loading favorites file: "/storage/.config/retroarch/content_favorites.lpl".
[INFO] [Playlist]: Written to playlist file: "/storage/.config/retroarch/content_history.lpl".
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[INFO] [Environ]: SET_GEOMETRY: 640x480, Aspect: 1.333.
[INFO] [KMS]: New FB: 1280x720 (stride: 5120).
[INFO] [Core]: Content ran for a total of: 00 hours, 01 minutes, 56 seconds.
[INFO] [Config]: Loading config: "/storage/.config/retroarch/retroarch.cfg".
[INFO] [Overrides]: Configuration overrides unloaded, original configuration restored.
[INFO] [Core]: No content, starting dummy core.
[INFO] [Core]: Content ran for a total of: 00 hours, 00 minutes, 00 seconds.
[INFO] [Core]: Unloading game..
[ERROR] [Environ]: GET_VARIABLE: reicast_texupscale - Invalid value.
[ERROR] [Environ]: GET_VARIABLE: reicast_texupscale_max_filtered_texture_size - Invalid value.
[INFO] [Core]: Unloading core..
[INFO] [Core]: Unloading core symbols..
[INFO] [Core]: Saved core options file to "/storage/.config/retroarch/config/Flycast/Flycast.opt".

shakkalakka, 2025-02-10:
Can bei closed. I switch back to an older Release without the issue.

closed 2025-02-10 by ToKe79