libretro/Lakka-LibreELEC #2040: xpi_gamecon: does not compile on devel branch

ToKe79, 2025-03-01:
We need updated source of this module used by RPi4-PiBoyDmg, as it does not build with 6.12.y:

```
make[1]: Entering directory '/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/linux-031a3826707c5a9c5761bb9c354756b6730051cf'
  CC [M]  /mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.o
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:123:6: warning: no previous prototype for 'gpio_func' [-Wmissing-prototypes]
  123 | void gpio_func(int pin, int state)
      |      ^~~~~~~~~
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:131:10: warning: no previous prototype for 'check_crc16' [-Wmissing-prototypes]
  131 | uint16_t check_crc16(uint8_t data[])
      |          ^~~~~~~~~~~
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:150:6: warning: no previous prototype for 'calc_crc16' [-Wmissing-prototypes]
  150 | void calc_crc16(uint8_t *data, uint8_t len)
      |      ^~~~~~~~~~
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c: In function 'gc_bcm_peri_base_probe':
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:394:19: error: implicit declaration of function 'of_find_node_by_path' [-Wimplicit-function-declaration]
  394 |         dt_node = of_find_node_by_path(path);
      |                   ^~~~~~~~~~~~~~~~~~~~
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:394:17: error: assignment to 'struct device_node *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  394 |         dt_node = of_find_node_by_path(path);
      |                 ^
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:400:13: error: implicit declaration of function 'of_property_read_u32_index' [-Wimplicit-function-declaration]
  400 |         if (of_property_read_u32_index(dt_node, "ranges", 1, &base_address)) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c: At top level:
/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.c:415:6: warning: no previous prototype for 'osd' [-Wmissing-prototypes]
  415 | void osd(void)
      |      ^~~
make[3]: *** [scripts/Makefile.build:244: /mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid/xpi_gamecon.o] Error 1
make[2]: *** [/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/linux-031a3826707c5a9c5761bb9c354756b6730051cf/Makefile:1937: /mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/xpi_gamecon-0.0invalid] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/mnt/hugo/BUILD/lakkabuildbot/Lakka-devel/build.Lakka-RPi4-PiBoyDmg.aarch64/build/linux-031a3826707c5a9c5761bb9c354756b6730051cf'
```

dmrlawson, 2025-03-02:
Experimental Pi (the authors/makers of the PiBoy) seem to have disappeared so I'm not sure we'll get an updated source unless someone is willing to fix it.

ToKe79, 2025-03-07:
temporarily (and maybe permanently) solved by using older (6.6.y) kernel for PiBoyDmg, but leaving the issue open

ShigeakiAsai, 2025-08-25:
Just information.

This build failure contains 3 errors and 4 warnings.
- each 3 errors are same cause.
  These errors are caused by the linux update.
  https://github.com/torvalds/linux/commit/ef175b29a242fea98f467f008237484b03c94834
  By this linux update, of_find_node_by_path() and of_property_read_u32_index() refferace are lost in xpi_gamecon.c.
  This errors are resolved by replace from "#include <linux/of_device.h>" to "#include <linux/of.h>".
- each 4 warnings are same cause.
  These warnings are resolved adding "static" each 4 functions in source file.

I confirmed to resolve build fail in devel branch, but I can't test it because I have no PiBoyDmg device.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/commits/xpi_gamecon-follow-raspberry_pi-linux-version-up/
Please wait a moment, I ordered it.

ShigeakiAsai, 2025-09-09:
I add the information.
I got PiBoyDmg device today.
It worked on devel branch with raspberrypi linux 6.12 and patched xpi_gamecon driver.

Please let me know if this is needed.

Note.
xpi_gamecon is not wireless controller driver, it is gpio controller driver.