libretro/Lakka-LibreELEC #2044: Kernel panic on 6.x-20250301-fff6e08

JulianGro, 2025-03-03:
trying to boot live or the installer of 6.x-20250301-fff6e08 results in a kernel panic. I tried both wayland and x11.
This is on an Asus Prime X299 Deluxe with Intel i7-7800x and an Nvidia GTX 1080.
Version 5.0 works on the same machine (thought after using it for a bit retroarch always segfaults now, but I guess that is unrelated). The machine itself generally works fine; I use it for VR on Windows.

[image]

ToKe79, 2025-03-03:
Hi, on first look it seems to be either corrupted download or flashing or medium. Can you try redownloading and re-flashing the image on a different medium? I would try myself, but I ran out of spare drives for testing. Thanks!

JulianGro, 2025-03-03:
I can't imagine it being corrupted, since I tried two different images for version 6.x-20250301-fff6e08. I made sure to `sync` and properly unmount Ventoy before disconnecting it from the computer that put the images on the drive. The drive itself should also physically be fine, since I use it all the time and version 5.0 works fine on that.
The dedicated USB-stick was using one of the images I copied to Ventoy as well.
So the options I see are:
- syslinux is broken but grub works
- the USB-drive that I use for Ventoy is causing issues (I had that on Debian Bullseye in the past. No kernel panic, but their installer couldn't deal with it for some reason.)
- the newer 6.x images generally don't boot on Ventoy while the old 5.0 images work fine.

ToKe79, 2025-03-03:
Try without Ventoy, it's possible that the kernel feature is not enabled to make Ventoy work (something with device-mapper iirc)

ToKe79, 2025-03-03:
and grub is now default for EFI boot, syslinux is for legacy boot. but it's possible that syslinux might be broken, try to switch you computer to legacy boot to confirm syslinux broken/working.

gouchi, 2025-03-04:
It seems we already  have [CONFIG_BLK_DEV_DM](https://github.com/libretro/Lakka-LibreELEC/blob/81c32c3f2fefa9024e6e69177d160ae53b2d65f0/projects/Generic/linux/linux.x86_64.conf#L2096) enabled in the kernel for Generic platform.

[Source](https://github.com/libretro/Lakka-LibreELEC/issues/1381)

ToKe79, 2025-03-04:
Therefore I think something on Ventoy side needs to be adapted, as we changed the EFI bootloader from syslinux to grub.

closed 2026-02-10 by ToKe79