libretro/Lakka-LibreELEC #2055: No Audio with rpi4 and Lakka 6.0

must4rd, 2025-03-19:
Hi there, 
i bought 2 8bitdo Ultimate 2C Wireless Controller and tried to setup a retro console. I checked Lakka 5 and didn't get the Controller to work. I downloaded the joypad conf and nearly tried everything i could with no success. It is still detected as a Mouse Input. 
In Recalbox, the controller is just working fine. 

So i tried lakka 6.0 and controller was working fine. But i do not have any audio output devices. 
Default Audio is alsathread.  I tried alsa and all given driver as well. 

A test to play some audio via aplay also failed.

Is it a known issue? Can i somehow get audio to work on lakka 6? or controller on lakka 5?

ShigeakiAsai, 2025-03-19:
Which audio interface device are you use?
HDMI, Headphone jack or other?

If you use HDMI, try to connect HDMI with MicroSD side please.

must4rd, 2025-03-19:
Oh sorry. It is HDMI.
What do you mean with MicroSD side? Im using a MicroSD Card ;-) 

aplay detected some devices. But in XMB GUI, there is no device available.

must4rd, 2025-03-19:
if you mean the HDMI next to MicroSD: I used this first and gave it a second try with the second hdmi.

ShigeakiAsai, 2025-03-19:
> if you mean the HDMI next to MicroSD:
Yes I mean. Sorry for my bad English.

The HDMI next to MicroSD can get audio on HDMI display with speaker.
More one, please push controller "start" button on XMB menu - Audio - Output - Device.
The audio device is reset.

If it can't resolve, let show which image version using please.

must4rd, 2025-03-23:
that was working! 
at least i reflashed the latest image used before and tried to configure audio first and now its fine! 
last time i configured services, games and other settings and then i ran into the audio problem!

Issue can be closed ;-)

closed 2025-03-24 by gouchi