libretro/Lakka-LibreELEC #2059: Failing to boot on Banana Pi M5 on 5.0 and Nightly 6.x

Gabriel-Arimatea, 2025-03-25:
If you are sure that it's a bug in Lakka, please answer the following questions:

- Which version of Lakka are you using?
    **5.0 and 6.x**
- What system hardware are you using?
    Banana Pi M5
- What did you do?
    Flashed the img into a microSD using etcher
- What did you expect to happen?
    Lakka to boot properly
- What happened instead?
    **mmc0: error -84 whilst initializing SD card** 
    **LABEL=LAKKA  Can't open blockdev** 
    **error in mount_flash: mount_common: could not mount LABEL=LAKKA**
- Description:
    I flashed lakka 5.0 and nightly 6.x versions into a microSD, both partitions are created correctly and with the correct names.
    When i turn on the BPI m5, it shows those errors and when I type exit, it shows the logo but still don't boot.

    Also the microSD was tested with other OS from BPI M5 and work normally.

    Am I missing something here or the image for BPI is really not working?

ShigeakiAsai, 2025-04-06:
I got same problem.
This problem also happens on LibreELEC 12.0.2 I think.

@Gabriel-Arimatea 
Could you please try LibreELEC 12.0.2 on your Banana Pi M5.
If this problem is also happened on your board, I will ask LibreELEC.

Thanks
ASAI, Shigeaki

chewitt, 2025-05-04:
@Gabriel-Arimatea please write this LE image to SD card and see if it boots okay on your board: https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.aarch64-12.80.2-bananapi-m5.img.gz

It contains this change https://github.com/chewitt/linux/commit/2cafd6c50f82584d0e0be3b5207523702a705e09

If you can confirm success I will send the patch upstream to the kernel.

ShigeakiAsai, 2025-05-05:
@chewitt 
This is my LE test result, Sorry for behavior report only (no log file)
Your fix is great!

LibreELEC - 5 microSD medias boot result on Banana Pi M5

a) ADATA  AUSDX64GUI3V30SA2-RA1 64GB-A2
- LE12-nightly -> Boot is Failed
= chewitt-12.80.2 => Boot is Passed, KODI is shown

b) HIDISC  HDMCSDX64GSW-WOA 64GB-A?
- LE12-nightly -> Boot is Failed
= chewitt-12.80.2 => Boot is Passed, KODI is shown

c) KIOXIA  LMEX1L064GG2 64GB-A?
- LE12-nightly -> Boot is Failed
= chewitt-12.80.2 => Boot is Passed, KODI is shown

d) SanDisk  SDSQUAB-O64G-GN6MN 64GB-A1
- LE12-nightly -> Boot is Failed
= chewitt-12.80.2 => Boot is Passed, KODI is shown

e) Transcend  TS64GUSD300S-A 64GB-A1
- LE12-nightly -> Boot is Passed, KODI is shown
= chewitt-12.80.2 => Boot is Passed, KODI is shown

Using image information.
- LE12-nightly -> LibreELEC-AMLGX.aarch64-12.0-nightly-20250416-e25b5e0-bananapi-m5.img.gz
= chewitt-12.80.2 =>  LibreELEC-AMLGX.aarch64-12.80.2-bananapi-m5.img.gz

So, I will try to Lakka-v5/6 build with your patch, and report.

Thanks
ASAI, Shigeaki

chewitt, 2025-05-05:
If it works with LE it will work with Lakka, thanks for reporting back 👍🏻

ShigeakiAsai, 2025-05-06:
@chewitt 
Thank you very much for your fix!
It works Lakka fine by your fix.

@Gabriel-Arimatea and all
If you want to confirm the fix this issue, please use my local build version.
- Lakka-v5.x version :
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/BananaPiM5-FixSdBootFail-Lakkav5
- Lakka-v6.x version :
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/BananaPiM5-FixSdBootFail-Lakkav6

I will send new PR for both versions later.
The Lakka devel branch version will be provided by LibreELEC future I think.

Thanks
ASAI, Shigeaki