libretro/Lakka-LibreELEC #2060: Switch - Cannot change time and date

ldroberts02, 2025-04-04:
I cannot change the time and date, other than the time zone. I've tried through hekate and switch OS, and it is required for my network's connections to initialize, else security checks fail for me with redirection errors.

gouchi, 2025-04-16:
You should try to provide some [log](https://lakka.tv/doc/Troubleshooting-Lakka/). 

@GavinDarkglider should have some insight.

GavinDarkglider, 2025-04-16:
> You should try to provide some [log](https://lakka.tv/doc/Troubleshooting-Lakka/).
> 
> [@GavinDarkglider](https://github.com/GavinDarkglider) should have some insight.

My 2 cents, don't have WiFi that depends on time/date to connect ..... That being said, plug into a computer via USB, use serial console to set date, then connect to WiFi.