libretro/Lakka-LibreELEC #2071: Lakka 6.x - Just an issue with PS2 games on LRPS2 core

Chris200418, 2025-05-03:
Hello.
When i started testing ps2 games i have noticed something.
In NTSC region games, it is working fine, but when i started playing a European region game (PAL), it showed a black bar up top.
For example, FIFA 2001 is playable in the NTSC version. 
However, on the PAL version it has that black bar which in some games barely shows some things, such as for example the score and the time limit.
Here are the differences that i have noticed:

[image]

[image]

Can this problem be fixed on the new lakka version? It would be really helpful.

My PC is a Lenovo Thinkcentre M93p with an Intel i7 4785t and Intel HD 4600 Graphics.

Thank you.

Ntemis, 2025-05-03:
HD4600 is EOL on mesa drivers.  Missing features and unfixable  with lakka. You have to report  directly to the lrps2 issues github

closed 2025-05-03 by Ntemis

Chris200418, 2025-05-03:
Ok.