libretro/Lakka-LibreELEC #2077: Add support for Intel Grphics

denser, 2025-06-03:
Hello, id of my mini pc videocard is
`pci/ven_8086&dev_46d4`
Could you please add it to supported?
Or any advice how to add it by myself?

I try to start any versions (5.x, 6.x, nightly) but no success. Device boots up but cant start retroarch service or xorg on x11.
I suppose the reason is unknown card id.

denser, 2025-06-03:
Btw. Live ubuntu successfully boot with good picture.

Ntemis, 2025-06-04:
Intel RSTe NVMe RAID Controller.

closed 2025-06-04 by Ntemis

denser, 2025-06-04:
> Intel RSTe NVMe RAID Controller.

[image]

denser, 2025-06-22:
Thx! @ShigeakiAsai !
It works! Strange I believe try that image))
With x11 - i have touch screen buttons on my TV - I copied /storage from my old laptop, there wasnt such thing. Is ot because x11 version?

denser, 2025-06-24:
Btw. With Generic - wont start retroarch. So need to use x11 one.

ShigeakiAsai, 2025-06-24:
I was able to start Generic.x86_64 retroarch on my N150 (GMKtec G3 Plus).
[image]

I am using Live USB booting with clean image.

Can you please share log file if you can get log?
Or, you can continue to use X11 as is if you feel no problem.