libretro/Lakka-LibreELEC #2084: Retroflag

andrewbean90, 2025-07-14:
The retroflag safe shutdown option barely works with Pi Case, and NESPi/ SNESPi cases. I have a PiStation (made out of the same hardware, and coding as the SNESPi & every other case by Retroflag), and when applying the Retroflag shutdown setting (and even after applying the Retroflag official optional lakka command for the safe shutdown script) it doesn't fully turn off the case. The fan in the case is still constantly running at full speed, and the Pi still receives power (as indicated by the red light on the Pi). I have a Raspberry Pi 4 (also called a Raspberry Pi 4 B+) 8GB model. In RetroPie, RecalBox, and Batocera the Pi actually shuts down completely but that is not the case in LAKKA. I should note that I have the latest "nightly" build circa May 09 2025.

ShigeakiAsai, 2025-07-15:
This is my local test version for Lakka-v6 RetroFlag case completely power off.
https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/Fix_completely_power_off_Lakka-v6.x_RETROFLAG_Pi_CASE

This test version will completely powered off when power switch is off.
But panic screen appears and does not powered off when it selects shutdown in Lakka menu without power switch off.

andrewbean90, 2025-07-15:
> Hello [@andrewbean90](https://github.com/andrewbean90)
> 
> The following are my understanding. Is it same as you?
> 
> * RetroPie
>   It is not included in official RetroPie image, it is provided by RETROFLAG.
>   https://github.com/RetroFlag/retroflag-picase
> * Batocera
>   It is not included in official Batocera image, it is provided by RETROFLAG.
>   https://github.com/RetroFlag/retroflag-picase
> * recalbox
>   It is included in official recalbox image.
> * Lakka
>   It is contained in Lakka's menu, but not fully function.
>   the shutdown is executed but power is not offed compleately when power switch is offed.
> 
> Thanks ASAI, Shigeaki

Yes the Pi doesn't completely shut off. The red light on the Pi is still on, and the fan in the case is still going at full speed.

andrewbean90, 2025-07-15:
> This is my local test version for Lakka-v6 RetroFlag case completely power off. https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/Fix_completely_power_off_Lakka-v6.x_RETROFLAG_Pi_CASE [...]
 
Tried that already, and it still doesn't work.

The red light is still on, and the fan in the case is still going at full speed.

andrewbean90, 2025-07-16:
> [@andrewbean90](https://github.com/andrewbean90) thank you for your confirmation. It seems you are still having problems, could you please reopen this issue?
> 
> Note. I ordered RetroFlag DREAM CASE/PiStation Case/SUPERPi 4 CASE today. I will try to fix this problem after I get these cases.
> 
> Thanks a lot ASAI, Shigeaki

For reference the red light on the Pi itself that indicates that the Pi is still on is what I am referring to when I said that the red light stays on. Not the red light on the case. The official Retroflag Safe Shutdown script has 3 scripts you can apply to either Batocera, RecalBox, or RetroPie. They also have 1 link to an approved Safe Shutdown script for LAKKA. However the LAKKA Retroflag option and even Retroflag's approved LAKKA script doesn't fully shut down the Pi as the red light on the Pi, and the fan in the case stay on. The other 3 Safe Shutdown scripts for Retroflag fully shut down the Pi.

andrewbean90, 2025-08-05:
> [@andrewbean90](https://github.com/andrewbean90) I have updated the RetroFlag safe shutdown a bit. https://github.com/ShigeakiAsai/Lakka-LibreELEC/releases/tag/RetroFlag-cases-completely-poweroff-002
> 
> I would appreciate it if you could check it when you have time.
> 
> Note. If you want to update Lakka system that using old Retroflag safe shutdown, after the update is complete, disable -> enable it via menu and reboot system please.

Doesn't work

[image]

ShigeakiAsai, 2025-08-10:
Thanks a lot

I don't know what is cause yet.
I need more time for investigation please.